analogdevicesinc/scopy #275: Inconsistent step size in frequency widget

tagoylo, 2017-09-27:
Scopy build: [54e3bff4](https://github.com/analogdevicesinc/scopy/commit/54e3bff48991ab0b36b01d2fb4b9febf35d2e0d3)

Open signal generator. Click on waveform.
Set frequency in fine interval, orange dot in the center.
Then try clicking on the +/- button, sometimes it has 2 unit interval, sometimes it has 1 unit.

amiclaus, 2017-10-02:
I have to mention that in my case, this issue is valid only for mHz values.

adisuciu, 2017-10-20:
This behavior usually occurs when the application is under heavy workload. I optimized the preview window with f2876e55e1d1ece4a1b2f3d358a623d9a0368e55 and f3e3f436ce07e6fc18038d2370f5f0a8163951df, so this should be fixed.

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the bug is fixed.

closed 2017-11-21 by amiclaus