analogdevicesinc/scopy #278: No unit specified in Y-axis of FFT

tagoylo, 2017-09-28:
Scopy build: [0408cc62](https://github.com/analogdevicesinc/scopy/commit/0408cc62f684672aa18bdf77f52b0dd7d72c1b9f)

As seen in the picture below, there is no unit specified in the y-axis of the FFT plot.
[image]

AlexandraTrifan, 2017-10-02:
Fixed with commit a71f88820779918610382b03927b3d9fe07bd7ac.

rgetz, 2017-10-02:
I would disagree - this isn't a bug.

Depending on the context, there shouldn't be numbers anywhere - just like other instruments:

[image]

[image]

[image]

[image]

All are oscilloscopes with a "FFT" view - no numbers anywhere. This is the expectation.

Spectrum analyzers are different, they have numbers.

[image]
No frequency numbers, but they do have amplitude numbers (and a single scale indicator).

[image]
same

[image]

same.

- so, there doesn't need to be units on each number when viewing an FFT from a scope view.

-Robin

AlexandraTrifan, 2017-10-02:
I understand. I will revert the previous commit.

AlexandraTrifan, 2017-10-03:
The FFT plot is back to its initial version through commit aa3aacd56687abbc10be61020985208cf80c0be5.

closed 2017-10-03 by AlexandraTrifan