analogdevicesinc/scopy #280: Spectrum Analyzer: Instrument doesn't automatically stop when m2k is disconnected

drbautista, 2017-10-02:
OS: Windows 7 64bit
 Scopy Version: 80794965
 Firmware Version: v0.14

Steps to reproduce:
1. Open and run the Spectrum Analyzer instrument.
2. Go to home and disconnect the m2k.

[image]

3. see that the run icon is still on for the Spectrum Analyzer.
4. When the m2k is reconnected the run icon is still present but the spectrum analyzer is stopped, but sometimes scopy crahes when reconnected in this situation.

dNechita, 2017-10-11:
Fixed with https://github.com/analogdevicesinc/scopy/commit/62bd02d123795fa55546a40a35a045bc79fa7ea1

amiclaus, 2017-10-11:
On Scopy build: 704cb7f (v0.8) the bug is fixed.

closed 2017-10-11 by amiclaus