analogdevicesinc/scopy #282: Burger menu enabled whenever other menu is enabled.

amiclaus, 2017-10-06:
Scopy version: 07380f6

I.e. select gear menu to open open the general settings pannel. The burger menu icon is simultaneously enabled. This happens also when opening other pannels like channel settings, measure, etc.
[image]

AlexandraTrifan, 2017-10-11:
The _settings_ button (burger menu icon) should be checked every time one of the following buttons is checked: measure button, cursors button, trigger button, channels buttons. This button handles the last settings panel that was opened.
The _general settings_ button (gear wheel) should disable all the other checked buttons. This button handles the general settings panel and should not interact with the _settings_ button.
This is fixed with commit 1a22c4adf897f968fb015cf9f4b3b6fd2e33dd53.

amiclaus, 2017-10-11:
On Scopy build: 704cb7f (v0.8) the bug is fixed.

closed 2017-10-11 by amiclaus