analogdevicesinc/scopy #283: Build failure due to incompatible signature of sr_period_string in libsigrok

balbirthomas, 2017-10-07:
Hi,

I did manage to build scopy on Debian GNU/Linux (Jessie version). I did use
  - libsigrok 0.5
  - libsigrokdecode 0.5
  - Qwt : experimental multiaxes branch (git sha  073145e7dba0665773329f214ab028b9f951baeb)
  - Qwtpolar version 1.1.1 with the qwtpolar-qwt-6.1-compat.patch (from the scopy mingw repository) applied
  - Gnuradio 3.7.10
  - Current git versions of iio and other Analog Devices code dependencies

However there was still a build failure in the Scopy file src/pulseview/pv/binding/device.cpp which seems to be using an older signature of the Sigrok string utility function sr_period_string . In particular the new version of this function requires an additional argument "precision". What should the value of this new argument be ?

AlexandraTrifan, 2017-10-16:
Hi,

As written in the Linux Build Guide (https://github.com/analogdevicesinc/scopy/wiki/Building-Scopy-on-Linux), Scopy used an older version of libsigrok. We added the necessary changes (commit 574c441ad3f1408f2fba5c034ceb7769f6220f64) to make Scopy compatible with the latest libsigrok. You should be able to build successfully the master branch now.

amiclaus, 2017-11-21:
I'll close this since the issue was solved.

closed 2017-11-21 by amiclaus