analogdevicesinc/scopy #286: Setting volts/div in oscilloscope doesn't respond correctly sometimes

tagoylo, 2017-10-11:
scopy build: [8f7e63f6](https://github.com/analogdevicesinc/scopy/commit/8f7e63f69fb815457566200f40c6619f6c861dbc)

Steps to reproduce: 

Select ch1.
[image]

Then click on the settings icon of ch2 (just the setting icon of ch2), the ch2 settings panel opens
Then change volts/div of ch 2
Select again ch1, the volt/div in ch1 also changes to the value set in volt/div in ch2
Then again select ch2, now the volts/div in ch2 is not changed to the new set value instead it is back to its previous value.
And volt/div in ch1 following the value of volts/div in ch2.

DanielGuramulta, 2017-10-12:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/af81c589f5ff78721fbe7c6a66b4ab684bee1e4c

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the bug is fixed.

closed 2017-11-21 by amiclaus