analogdevicesinc/scopy #289: Voltmeter:  Unstable when one channel measures DC and the other measures AC (800Hz - 40kHz)

drbautista, 2017-10-12:
OS: Windows 7 64bit
 Scopy Version:  290a894c 
 Firmware Version: v0.14

Steps to reproduce:
1. Open Voltmeter instrument and set one channel to DC and the other to AC (800Hz - 40kHz)
2. Connect the power supply to the channel in DC Mode and set any voltage.
3. Run first the voltmeter then the power supply.
4. Notice that the voltmeter won't be able to read the voltage input or there will be a huge delay before it settles to the right value.
5. If it managed to read the right value after some time. Try to adjust or turn off the power supply, the reading will not be updated.
6. The value will only be updated quickly if the AC (800Hz - 40kHz) mode will be changed.

*This happens on both channels.

AlexandraTrifan, 2017-10-27:
I managed to reproduce this with the provided Scopy Version. This should be fixed after commit d65d1a764c578ebf9aa89f385467289c6f0f40ef.

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the bug is fixed.

closed 2017-11-21 by amiclaus