analogdevicesinc/scopy #291: Spectrum Analyzer: Scopy crashes when the unit is changed

drbautista, 2017-10-13:
OS: Windows 7 64bit
 Scopy Version:  ec8763f3 
 Firmware Version: v0.16

Step to reproduce:

1. reconnect m2k device so that there's no signals on the spectrum analyzer or reopen Scopy.
2. go to spectrum analyzer instrument and go to sweep settings. [do not run spectrum analyzer]
3. change the unit from dbFS to any available unit. 
4. Scopy will crash.

[image]

*Scopy will not crash if the unit is changed when the spectrum analyzer has been run.

dNechita, 2017-10-13:
Fixed with: https://github.com/analogdevicesinc/scopy/commit/6ba153bbbc3c6c7d230ee53af5684de335245999

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the bug is fixed.

closed 2017-11-21 by amiclaus