analogdevicesinc/scopy #292: Auto Trigger mode shows different behavior when FFT is used

tagoylo, 2017-10-13:
Scopy build: [290a894c](https://github.com/analogdevicesinc/scopy/commit/290a894c4e38e3b47c898af74593ae8d65f93181)

Steps to reproduce

In signal generator: 1kHz square wave, 5V

Oscilloscope: 1ms/div, 1V/div
Trigger mode: Auto

Open FFT plot.
FFT size is 1024.
Run instrument.

Then change FFT size,  16384 and above. The scope trace label shows "Auto", unless the user change the trigger mode to Normal, the label is "Triggered".
[image]

Stop instrument.
Turn off FFT. Then run again the oscilloscope, Auto mode, the label now shows triggered.
[image]

Then turn on again FFT. FFT size is the same as when closed, 16384. Run instrument. The label now shows "Triggered".
[image]

AlexandraTrifan, 2017-10-24:
Fixed with commit 4cb208bfac1831ea341a1963492bc8f8e393627a.

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the bug is fixed.

closed 2017-11-21 by amiclaus