analogdevicesinc/scopy #294: Run button also scrolls up

tagoylo, 2017-10-16:
Scopy Build: [33b0b388](https://github.com/analogdevicesinc/scopy/commit/33b0b388d6f39486ad1ff360da4a0078aef89dd4)

When in minimum window and the user scrolls down up to the data logging and Timer section, the Run button is nowhere to be seen unless the user scrolls back up again. 

[image]

I think it would be convenient for the user to leave the Run button fixed instead of it being part of the scrollable section when in minimum window size.

AlexandraTrifan, 2017-10-16:
Fixed with commit aa4b1d4f48822a3e5f6ccdd7ca637f165a29b13e.

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the bug is fixed.

closed 2017-11-21 by amiclaus