analogdevicesinc/scopy #299: Distorted waveform is generated

tagoylo, 2017-10-17:
Scopy build: [574c441a](https://github.com/analogdevicesinc/scopy/commit/574c441ad3f1408f2fba5c034ceb7769f6220f64)

Steps to reproduce:
Set signal generator to the following:
sine wave, 10 V Amplitude
Monitor in oscilloscope.
The picture below is the result.
[image]

Also monitored it with Discovery and the result is the same.
[image]

The distortion happens when amplitude is set to 10 V and frequencies below 1kHz. Set signal generator frequency in fine mode.

adisuciu, 2018-01-12:
Fixed with f044b3c58b3c40dd54e3ffb0c3581aeb883282f8

amiclaus, 2018-01-16:
On Scopy build e9a6c85 (v0.10) the bug is fixed.

closed 2018-01-16 by amiclaus