analogdevicesinc/scopy #300: Math equation in Math Channel Settings Panel

amiclaus, 2017-10-19:
In the Math Channel Settings Panel there should be displayed the equation used to generate the signal and  the user to be able to edit it. In this manner, the user can easily make a correspondence between the displayed signal that was generated and the input that made the generation. Being able to edit the math function and apply it within the same channel will avoid the situation when the equation was initially written wrong and needs adjustment (now the user has to delete the math channel and create a new one, writing again the whole equation).

DanielGuramulta, 2017-11-15:
Implemented with commit: https://github.com/analogdevicesinc/scopy/commit/e56cb838bc229c3e3042679c6a2cfcea6b40103d

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the enhancement is implemented.

closed 2017-11-21 by amiclaus