analogdevicesinc/scopy #313: Run/Stop button not responding properly

tagoylo, 2017-11-13:
Scopy build: [5ae94afb](https://github.com/analogdevicesinc/scopy/commit/5ae94afbe182e0df3e535786d1b31e04b58fb815)

Steps to reproduce:
Connect DIO0 to DIO1
Set DIO0 as output and DIO1 as input. 
Run Digital IO using the white box in the instrument list.
Result: The Run/Stop button at the upper right corner of the interface is not responding properly. It should appear as an orange button witjh stop label. As seen in the picture it has "Run" label instead. 
And when changing the logic state of the output, the channel does not respond.
[image]

Stop the instrument using the button in the instrument list.
Now, run again the instrument using the Run/Stop button at the upper right corner of the interface.
The stop the instrument using the button in the instrument list.
Result: The run/stop button at the upper right should appear as green with the label of "Run", but as shown in the picture it is instead with "Stop" label. 
And then when changing the logic states, it still responds in which it should not
[image]

DanielGuramulta, 2017-11-13:
Fixed with commit : https://github.com/analogdevicesinc/scopy/commit/48f7a62c9286580ea28e9cf312df897ac49133d0

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the bug is fixed.

closed 2017-11-21 by amiclaus