analogdevicesinc/scopy #314: One period not displayed at max frequency

amiclaus, 2017-11-21:
Scopy build: c8ad583
Use only CH1. (Disable CH2)
Use Square Wave.
Start i.e. from 1kHz frequency.
Press "+" button for the frequency until reaches max frequency. At 37.5MHz on the plot there is not displayed an entire period.
[image]

adisuciu, 2018-01-19:
Fixed with  b33fb00

amiclaus, 2018-01-19:
The bug still persists on 92484c8 v(0.10).

Start from 1kHz, and keep the "+ button pressed" until reaches 37.5MHz.

Sometimes, the procedure must be repeated to see the bug.

[image]

adisuciu, 2018-03-22:
Fixed with  3406832

amiclaus, 2018-03-23:
On Scopy build: 08b2360 the bug is fixed.

closed 2018-03-23 by amiclaus