analogdevicesinc/scopy #316: Signal Generator: Signal distorted when single button is pressed on the oscilloscope

drbautista, 2017-11-22:
OS: Windows 7 64bit
Scopy Version: c8ad583
Firmware Version: v0.16

Steps to Reproduce:
Apply the following setting  on the Signal Generator:
Channel 1: Sinewave
Amplitude: 5V
Frequency: 200Hz

Channel 2: Falling Ramp Sawtooth
Amplitude: 9V
Frequency: 300Hz

Connect Oscilloscope 1+ to AWG1 and Oscilloscope 2+ to AWG2.
Connect Oscilloscope 1- and 2- to GND.

Run Signal Generator and repeatedly use single capture on oscilloscope. 
it will capture the distorted part of the signal as show in the picture below. it happens on both channels and different waveforms.(It is easier to reproduce on low frequencies)

[image]

[image]

Also, to show that this is not an Oscilloscope issue. 
I tried capturing it on different scope. 
It also occurs even if the M2k's oscilloscope probes are not connected on AWG1 or AWG2.

[image]

acostina, 2017-12-04:
I was able to reproduce the issue with the latest scopy, but if the data capturing is done through iio-oscilloscope (connected through usb-ethernet) instead of scopy , the glitch does not appear, even if the adc data is captured at 100 MSPS and a high number of samples.

adisuciu, 2017-12-04:
Also when generating data outside of Scopy (using a simple libiio program), and data is captured with the "Single" button in the oscilloscope, two small 100ns glitches appear with 10 ms distance in between them. It looks like the timing is the same as the glitches found in the original comment however the actual glitch time is a lot smaller.

lclausen-adi, 2017-12-07:
Should be fixed by https://github.com/analogdevicesinc/linux/commit/5b3ea27b2b74ac30623f38075533447f0028c30e

amiclaus, 2018-03-12:
On Scopy build: 6bd7075 (v0.11) the bug is fixed.

closed 2018-03-12 by amiclaus