analogdevicesinc/scopy #317: Voltmeter: AC Mode at high frequency (800Hz to 40kHZ) stucks up when the other channel is on DC or low frequency AC mode

drbautista, 2017-11-22:
OS: Windows 7 64bit
Scopy Version: c8ad583
Firmware Version: v0.16

Steps to Reproduce:

Set voltmeter's channel 1 to AC (800Hz to 40kHZ mode)
Set voltmeter's channel 2 to DC Mode

Connect 1+ to AWG1 and 1- to GND.
Connect 2+ to Vp and 2- to GND.

Set Signal Generator's channel 1 setting to:
Amplitude: 5V
Frequency: 1kHz
Set positive Power Supply to 5V.

**First** Run voltmeter instrument
Then run signal generator and Power Supply instrument.

Notice that channel 1 is on pause while channel 2 started reading the 5V supply.

[image]

channel 1's reading will be updated when you stop voltmeter and run it again.

This behavior also happens when channel 2 is set to AC(800Hz - 40kHZ) while channel 1 is on DC or AC(20Hz - 800Hz), in this scenario, channel 2 is the one that is not functioning properly.

AlexandraTrifan, 2017-11-22:
This bug was introduced in the latest commit on the voltmeter. Should be fixed after commit c05c1016b9e2cf8342784a09330ffc416820dd5b.

amiclaus, 2017-11-24:
On Scopy build: 3aff056 (v0.9) the bug is fixed.

closed 2017-11-24 by amiclaus