analogdevicesinc/scopy #320: Spectrum Analyzer: Marker data on table missing when m2k is disconnected

drbautista, 2017-11-27:
OS: Windows 7 64bit
Scopy Version: 3b12fc9
Firmware Version: v0.16

Steps to reproduce:

Open and Run Spectrum Analyzer instrument.
On channel 1 or 2, activate any number of markers.
Enable Marker table.
Disconnect and reconnect m2k from the Home Menu.

[image]

Rerun Spectrum Analyzer instrument and notice that the markers are still active on the plot window but the information on the marker table are  not present.

claudiupop, 2018-03-15:
Fixed with 8574df18c4a3a53f37041bd0adc0cc6eb1d1323d

amiclaus, 2018-03-16:
On Scopy build: 1e62479 (v0.11) the bug is fixed.

closed 2018-03-16 by amiclaus