analogdevicesinc/scopy #322: No limit in adding the same device through ip address

tagoylo, 2017-12-06:
Scopy build: [64f6a3d1](https://github.com/analogdevicesinc/scopy/commit/64f6a3d10301d38cf66171093c15fd500fe4418f)

Steps to reproduce:
Click the "Add +" to add device through ip address.
Input the ip address of the device.
After adding the said device, repeat previous steps to add again the same device through ip address.
One will still be able to add the device though it is already added. It is somewhat redundant. There is no limit in adding the same device.

[image]

DanielGuramulta, 2017-12-08:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/15cb0e05861abdd02dbb65066493d0ffdd070124

amiclaus, 2018-01-16:
On Scopy build e9a6c85 (v0.10) the bug is fixed.

closed 2018-01-16 by amiclaus