analogdevicesinc/scopy #328: Create Linux packages

AlexandraTrifan, 2018-01-12:
Since Scopy is building on Travis-CI, we need to create packages. (.deb, .rpm)

rgetz, 2018-01-12:
how easy will this be, since we are using patched versions of all the prerequisites?

AlexandraTrifan, 2018-01-12:
I am still working on finding a good approach. One option would be to statically link some of the prerequisites (i'm not sure if it's recommended). Or maybe we could create .deb packages for the patched dependencies, and use those.

Another idea I found is AppImage. (https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages). 
If we would want to use this, it should work on all the common Linux distributions.

balbirthomas, 2018-04-30:
Hi,
I am trying to create a deb source package for scopy. I already have source packages for current git versions of its dependencies (listed below) to build against

1) libiio
2) libad9361-iio (Current official debian version is much too old)
3) gr-iio
4) libqwt-qt5-dev (experimental multiaxes branch)
5) libqwtpolar-dev (>= 1.1.1)

Further the following dependencies are already available as deb packages for Debian/Ubuntu

1) libglibmm-2.4-dev
2) libsigrok-dev (>= 0.5)
3) libsigrokcxx-dev (>= 0.5)
4) libsigrokdecode-dev (>= 0.5)
5) libvolk (1.3)
6) gnuradio (3.7.10)

The versions refer to current Debian/Stable (Jessie) on which I am building. The sigrok package is a backport (no changes necessary) to Debian/Stable from Debian/Unstable. Only scopy itself remains to be packaged. However the trouble I am having is collating a well defined list of scopy's dependencies and their versions (official or patched) and keeping this information up to date. For instance I was able to build scopy with above dependencies prior to January 2018 (using cmake build system not deb). However changes since fail to build. For example : 
```
In file included from /home/bt/docs/lab/tools/Analog Devices ADALM/git/scopy/src/tool_launcher.hpp:43:0,
                 from /home/bt/docs/lab/tools/Analog Devices ADALM/git/scopy/src/debugger.h:35,
                 from /home/bt/docs/lab/tools/Analog Devices ADALM/git/scopy/src/debugger.cpp:20:
/home/bt/docs/lab/tools/Analog Devices ADALM/git/scopy/src/signal_generator.hpp:77:16: error: ‘GR_TRA_WAVE’ is not a member of ‘gr::analog’
  SG_TRA_WAVE = gr::analog::GR_TRA_WAVE,
                ^~
CMakeFiles/scopy.dir/build.make:1048: recipe for target 'CMakeFiles/scopy.dir/src/debugger.cpp.o' failed
```
This seems like a requirement on a newer version of ```gr-analog``` from the gnuradio project. Please correct me if I am mistaken here.  I am unable to determine which version of gr-analog (gnuradio) this requires. 

In view of the above it will be very useful to have a list of dependencies and versions in the README.md in the this git repository. Also it will make maintaining the source package (build dependencies) much easier if any custom patches (libqwt-qt5-dev multiaxes ?) can be upstreamed and mainlined. Further if conditional compilation could be supported for features not available in older dependency packages, building on current official distribution releases, may become easier without having to wait for their next release. This is presuming, the desire is to obtain build dependencies, as much as possible from official distribution packages so that some day have scopy can be available from the distribution's own package repositories (which would be the most convenient option for those distribution users). Till such time it may be possible to create a convenient apt-get'able repository for Debian/Ubuntu for instance using ```reprepro``` once source and binary packages can be built on those respective distribution versions, in order to minimize replacing the user's installed official dependency packages with non official ones as much as possible.

I would be grateful for any help. I can upload current package metadata (debian sub-directory) to github if desired.

balbirthomas, 2018-05-02:
Good News. Scopy builds and compiles easily on Debian/Stretch with 

1. Three packages from Debian Testing/Unstable
     - libiio
     - gr-iio (libgnuradio-iio0)
     - libad9361

2. Two patched Debian package
    - gnuradio 3.7.10 (with patches of Adrian Suciu )
    - libqwt 6.1 (with multiaxes patch)

3. A custom Debian package for Qwt Polar (not part of Debian)

All the other dependencies are available within Debian Stretch (and later versions). Scopy itself was easy to package. However I am unable to fully test since though since scopy runs and detects my ADALM2000 but it is not able to "identify" the device and asks for a firmware update for which I am lacking instructions.

rgetz, 2018-05-02:
Latest release requires a cal file for the power supplies to work.

AlexandraTrifan, 2018-05-03:
The "Identify" functionality will not work with the latest M2K firmware release (https://github.com/analogdevicesinc/m2k-fw/releases), because this will be added in the next release. But you should be able to use Scopy with ADALM2000. You can connect to the board and use the instruments without the "Identify" functionality (which would only blink a led on your board).

mhennerich, 2018-05-03:
I will create an new release today or tomorrow. Things got a bit delayed since we want to create a new libiio release prior in releasing a new m2k firmware.

mhennerich, 2018-05-03:
[m2k-fw-v0.19-1-g7ce2-dirty.zip](https://github.com/analogdevicesinc/scopy/files/1970576/m2k-fw-v0.19-1-g7ce2-dirty.zip)

Here is a firmware build that should have all the required features

balbirthomas, 2018-05-03:
Thank you for your help. I can confirm that with firmware 0.19 I am able to connect with device, generate a signal and capture in the oscilloscope. (Still same problem with identify - Not an issue that bothers me)

For whatever it is worth I am attaching two zip files that contain the "debian" folders for building qwtpolar and scopy debian packages. As mentioned above all others are already packaged for Debian/Ubuntu. Please replace "NAME" and "EMAIL" in Changelog, control, copyright, README and scopy.man if you use them. The build depends are intentionally broad and generic, you may want to fine tune. 

Attach: 
[scopy-debian.zip](https://github.com/analogdevicesinc/scopy/files/1972404/scopy-debian.zip)
[qwtpolar-debian.zip](https://github.com/analogdevicesinc/scopy/files/1972405/qwtpolar-debian.zip)

rgetz, 2018-05-04:
Thanks for your help / assistance.

adisuciu, 2021-11-16:
That's awesome !

We can probably create releases for the packages that are missing them. However latest upstream for qwt is a problem because upstream doesn't include stuff from qwt-multiaxis branch which we use. The maintainer of qwt mentioned that the mainline and this branch will go in parallel for a while.

As for gnuradio, we will probably wait for gnuradio 3.10, and then port it.

-Adrian