analogdevicesinc/scopy #330: Oscilloscope: Other configurations doen't load correctly from saved profile

tagoylo, 2018-01-15:
Scopy Build: [3d8729db](https://github.com/analogdevicesinc/scopy/commit/3d8729dba61a68a2bc7d9edd0becd6b80a1b8063)
firmware version: v0.17

Steps to reproduce:

Save the following configurations as profile1.ini

Enable FFT, XY plot, labels of plot
Change channel thickness to 5
Enable software AC Coupling for Ch1
Enable cursor, change transparency to 50%, change position to lower right corner

after saving, close scopy and delete Scopy.ini in C:\Users\your_username\AppData\Roaming\ADI

Open Scopy, and load profile.ini

The result, there are configurations that doesn't respond/load, such as the ff:
FFT
XY plot
Export All
label preference
thickness
cursor readouts
software AC coupling
orange dot for fine/coarse interval

AlexandraTrifan, 2018-01-18:
The missing properties for the Oscilloscope such as whether FFT and XY views are enabled, export all button is checked, cursor readouts position and transparency and AC coupling were added with commit 17b3be145a9dfadfb7e1d729c3afc30d8859ab83.  I couldn't reproduce the behaviour of channel thickness not being loaded. The preferences, such as labels enabled on the plot, will be saved/loaded from a separate configuration file.

amiclaus, 2018-01-18:
On Scopy build: 62e9f54 (v0.10) the bug is fixed.

closed 2018-01-18 by amiclaus