analogdevicesinc/scopy #331: Power Supply: Orange dot for fine/coarse interval doesn't respond to what is saved in profile

tagoylo, 2018-01-15:
scopy build: [3d8729db](https://github.com/analogdevicesinc/scopy/commit/3d8729dba61a68a2bc7d9edd0becd6b80a1b8063)
firmware version: v0.17

steps to reproduce:

Set positive output as 5V
Then save as profile_ps.ini

Then change configurations,
set to fine interval, with orange dot in center,
change to tracking 50%
set 3V positive output

after setting these configurations, load the profile_ps.ini file 

As seen, the orange dot for fine/coarse interval is still there in which it should not. Since in profile_ps.ini we set it only to coarse interval. Other than that other configurations load properly.

DanielGuramulta, 2018-04-20:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/f8283e27b9e8afafad2e0b1f9786675b90f94011

amiclaus, 2018-05-02:
On Scopy build: 87e17a9b the bug is fixed.

closed 2018-05-02 by amiclaus