analogdevicesinc/scopy #336: Scopy version does not update when a previous version is already installed

tagoylo, 2018-01-16:
Installed [3d8729db](https://github.com/analogdevicesinc/scopy/commit/3d8729dba61a68a2bc7d9edd0becd6b80a1b8063)

Then install new build without uninstalling previous build.

After installation and reboot, the build/version seen on the tab is still the previous version, also when exporting from oscilloscope the scopy version indicated in the file is still the previous version.

But it updates when completely uninstalling scopy and installing new version. 
[image]

AlexandraTrifan, 2018-01-18:
After commit cf5c2823e78d1713ced45cfe20e664bab1f40752 Scopy installer should replace any previous version with the new version.

amiclaus, 2018-01-19:
The bug is fixed.

closed 2018-01-19 by amiclaus