analogdevicesinc/scopy #338: Noise signal visibility changing based on sample rate

amiclaus, 2018-01-16:
Scopy version: e9a6c85

Connect AWG1 to channel 1 of the Oscilloscope
Generate a 2V amplitude and 2kHz frequency Sine wave with Uniform Noise Type of 1V Amplitude:
[image]
View the signal on channel 1 of the Oscilloscope.
By changing the Time Base (implicitly the Sampling Rate) the Noise applied is attenuated, being less visible.
[image]
[image]
[image]

I don't think we want this things to happen for this particular setup.

Are there any solutions for this situation?

damercer, 2018-01-16:
The digital filtering implemented in the M2K to decimate the sample rate will filter out the higher frequency components of the noise which I assume has a BW equal to 1/2 the 75 MHz AWG base sample rate? If you looked at the AWG output with a real digital scope you would not see this effect ( unless you turned on the digital filtering option in the scope ).

M2K software should implement ( allow ) simple decimation where samples are just dropped as an option rather using digital filtering. Then this will effect not happen. And / Or you could allow the user to select the amount of decimation filtering vs sample dropping independent of that automatically chosen by setting the time/base.

Doug

adisuciu, 2019-02-27:
Added option to remove HW filtering with [623cf33](https://github.com/analogdevicesinc/scopy/commit/623cf33329fdec34c5428b71910bcb2745ae6cea) .

Issue further tracked here:https://github.com/analogdevicesinc/scopy/issues/506

closed 2019-02-27 by adisuciu