analogdevicesinc/scopy #339: Uniform Noise Amplitude meaning

amiclaus, 2018-01-16:
Scopy version: e9a6c85

For Uniform Noise Type, currently the amplitude is implemented as peak value.

[image]

Shouldn't it represent the peak-peak value?

damercer, 2018-01-16:
Noise with a uniform distribution i.e. equal probability of having any value within the range of the distribution, will have a p-p vale equal to the range ( assuming infinite bandwidth of course ). Other probability distributions like Gaussian ( often used to model thermal noise ) will have different p-p values.

Doug

adisuciu, 2018-01-16:
When implementing the noise source, I used the gnuradio noise_source block. 
https://gnuradio.org/doc/doxygen/classgr_1_1analog_1_1noise__source__f.html#a2165ba3c7cc54a1dcb8260e318976dba

It is mentioned that the amplitude parameter, is actually the standard deviation. In order to make Scopy more user friendly we can scale the amplitude given to the block, so that the noise is within range.

damercer, 2018-01-16:
If the function used to generate the noise has a "standard deviation" 
then it is by definition not uniform ( i.e. most likely Gaussian ). So 
you need to call it what it is.

Gaussian noise.

If you want Uniform distribution noise then you need to use a pure 
Random number generator with uniform distribution.

Doug

adisuciu, 2018-01-16:
There are multiple types of noise implemented by gnuradio, you can select them from the dropdown in Scopy.
The thing is that if we rename the amplitude to "standard deviation" it might confuse the average user.  Do you think it's worth renaming the control based on what kind of noise we are generating ? 

Uniform - Amplitude
Gaussian/etc - Standard Deviation

damercer, 2018-01-16:
That's a good question. Users are probably thinking in terms of the apparent amplitude, noise voltage or noise power (nV/rtHz sort of thing) of the signal not some mathematical construct. So I think the way that would make the most sense from the users perspective would be for them the specify the type of noise ( distribution function or shape ) and the noise spectral density. Does that make sense?

A related question is should there be a BW or frequency spectrum distinction like white noise vs pink noise or 1/F noise etc.

Doug

rgetz, 2018-01-16:
a few examples;

https://literature.cdn.keysight.com/litweb/pdf/5991-0692EN.pdf?id=2202606
Variable-bandwidth noise

https://literature.cdn.keysight.com/litweb/pdf/33500-90901.pdf?id=2197440
is described as "Quasi-Gaussian Noise" (p 180).  All that is adjusted is bandwidth. (which also effects amplitude).

adisuciu, 2018-01-22:
I know it's not 100% correct, but I implemented the following gnuradio flow, to minimize confusion:
[image]

ui_ampl is the set amplitude in the UI
value is a value (that was found experimental) that generates noise within the amplitude range. The rail block just makes sure there are no samples outside of range.

A few examples
[image]
[image]
[image]

I know it's not statistically correct, but the question is, is it good enough ?

rgetz, 2018-01-22:
For sure - any sort of instrument can not do a true Gaussian distribution, there is zero
probability of getting a voltage beyond the instrument’s Vpp setting. I think this makes sense.

It really is determined by if "Amplitude" is peak (which it looks like in the figures), or RMS. The crest factor (peak voltage divided by RMS voltage) should be approximately 4.6 for signals like this.

I guess this brings up two things for me (which may be future feature requests), 
 - is it possible to see the waveform preview in the frequency domain (in a tab or something?) That might make it easier for people to understand what they are getting.
 - do we measure crest factor in the scope measurements?

adisuciu, 2018-01-22:
- It's not possible to view the frequency domain of the signal, what you can do is create a loopback and run the spectrumanalyzer/scope in FFT mode to view the frequency domain. However the signal that is displayed is not the same signal that is generated. When clicking play, a new buffer is generated, and M2k works in  buffer mode (the same buffer is repeated over and over again). Therefore the noise is the same within a buffer, but different every time we hit the play button (and everytime we preview/make changes to the signal).

- We do not measure crest factor in scope measurements.

If we just want to study what happens, we can always create a gnuradio flow with an fft measurement and analyze the characteristics of the noise signal generator.

damercer, 2018-01-22:
I don't know anything about what GNU radio is doing when it generates a 
noise signal but wouldn't the spectrum of the noise just be flat within 
the bandwidth based on the sample rate? And just some flat noise floor 
with say a single frequency sinewave superimposed if that's what you are 
constructing. Not sure how displaying the spectrum of the noise provides 
any more information.

If the Scope tool already calculates the RMS and p-p ( Max - Min ) 
values then calculating the crest factor is simple. Does the Scope tool 
have a way for the user to do simple math on any ( or all ) of these 
pre-calculated measurements and display the results?

Doug