analogdevicesinc/scopy #341: Signal Generator: Graphical Representation unstable at high frequency setting

drbautista, 2018-01-17:
OS: Windows 10 64bit
Scopy Version: d5f3f32
Firmware Version: v0.17

Steps to reproduce:

1. Open Signal Generator Instrument
2. Set both channels' to waveform type and set both to 10V.
3. Set both channels' frequency to 2MHz or higher.
4. The graphical representation of the set waveform is unstable. 

[image]

The image shows that channel 2 is set to Sinewave but the graphical representation shows falling ramp sawtooth.

adisuciu, 2018-01-19:
fixed with  b33fb00

amiclaus, 2018-01-19:
On Scopy build (v0.10) 92484c8 the bug is fixed.

closed 2018-01-19 by amiclaus