analogdevicesinc/scopy #342: Digital IO: Group/individual configuration does not respond to saved profile configuration

tagoylo, 2018-01-17:
Scopy version: [d5f3f32d](https://github.com/analogdevicesinc/scopy/commit/d5f3f32d1c9f679850d286116ebbb9c5bad79f16)
firmware version: v0.17

Steps to reproduce:

Set the following configuration

Set DIO0-7 as group, set as out
and DIO8-15 as group, set as in

save profile as dio_a.ini

set DIO0-7 as individual pins, same as DIO8-15. Change other configurations.

Then load dio_a.ini

The result is not the same as what is saved in profile, it doesn't set the pins as grouped.

DanielGuramulta, 2018-01-17:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/cb1ff7baf670b74cfa56609c0ae6d6530de958c8

amiclaus, 2018-01-18:
On Scopy build: d85de1c (v0.10) the value set for the DIO out group is not always saved properly (if you change to invidiual DIO after saving the session and then loading session).

DanielGuramulta, 2018-01-18:
This commit( https://github.com/analogdevicesinc/scopy/commit/38b13b8192cea3e478777ed5c5e000cb22d7cfbc ) fixes the above mentioned issue.

amiclaus, 2018-01-18:
On Scopy build: 62e9f54 (v0.10) the bug is fixed.

closed 2018-01-18 by amiclaus