analogdevicesinc/scopy #343: Pattern generator: Other configurations doesn't respond to the saved profile

tagoylo, 2018-01-17:
Scopy version: [d5f3f32d](https://github.com/analogdevicesinc/scopy/commit/d5f3f32d1c9f679850d286116ebbb9c5bad79f16)
firmware version: v0.17

The following configurations doesn't respond to what is saved in the profile:

Hide inactive - when loading a profile that involves hidden channels, the channels doesn't show hidden after loading

output, PP or OD - this also does not respond to what is saved

thickness - when loading a profile that has been saved with thckness other than 1, the thickness that shows is 1 and not the one that is saved

Color Settings - when saved with other colors other than default, the colors that shows when loaded is not the one that is saved

adisuciu, 2018-01-19:
Fixed with a58f46cd929af22b32cbf1925acc8076a15e337a

amiclaus, 2018-01-19:
On Scopy build (v0.10) 92484c8 the bug is fixed.

closed 2018-01-19 by amiclaus