analogdevicesinc/scopy #346: saving/loading with thickness configuration doesn't respond when channel settings is open

tagoylo, 2018-01-19:
Scopy build: [62e9f549](https://github.com/analogdevicesinc/scopy/commit/62e9f5493dc2dffed72c1a02bb173a0f2d86ab4c)
firmware version: v0.17

This is in reference to item [#330,](https://github.com/analogdevicesinc/scopy/issues/330) in reproducing the bug regarding thickness.

Steps to reproduce:

save a profile with thickness 5, profile1.ini
change the thickness to other number
Load profile1.ini
thickness doesn't change
close ch1 settings
the open again ch1 settings
Now the thickness changed

when ch1 settings is closed and load profile1.ini, thickness changes
Only when channel settings is open, thickness don't respond/change

AlexandraTrifan, 2018-01-19:
Fixed with commit d17e9c1061496bf51dbe2f419ef4ddd83ac87d09

amiclaus, 2018-01-19:
On Scopy build (v0.10) 92484c8 the bug is fixed.

closed 2018-01-19 by amiclaus