analogdevicesinc/scopy #354: Signal Generator: Waveform output clipped at 10Vpp amplitude

drbautista, 2018-01-24:
OS: Windows 10 64bit
Scopy Version: 32f549d
Firmware Version: v0.17

Steps to reproduce:
1. Open Signal Generator instrument and set either channel's settings to:
         Waveform Type: Sinewave
         Amplitude: 10V
         Frequency: 500kHz
         Offset: 0V
         Phase: 0 degrees

2. Disconnect M2k from the home panel
3. Disconnect M2k from the PC or laptop
4. Reconnect M2k and run Signal Generator and oscilloscope.

[image]

The waveform will be clipped.
The clipping problem may sometimes be solved by simply reconnecting the M2K from home panel or restarting the Scopy to recalibrate the instrument.

adisuciu, 2018-02-05:
Cannot reproduce

adisuciu, 2018-03-13:
Retest this with new boards please.