analogdevicesinc/scopy #359: Push-pull/Open Drain issue

amiclaus, 2018-02-06:
Scopy version: 32f549d

It seems that output signals are always working in Push-Pull mode, whether you set the output of the pattern to OD or PP.

acostina, 2018-02-07:
Fixed in HDL: https://github.com/analogdevicesinc/hdl/tree/hotfix_m2k_2017_r1 
However, it appears that software doesn't write the corresponding bit from register 0x40 to 0 when push-pull is selected, nor does it apply the push-pull/open-drain setting change in real time.  The change is applied only if other settings are modified, or stop/start is pressed

adisuciu, 2018-02-07:
Fixed with commit - https://github.com/analogdevicesinc/scopy/commit/7a179c82b6acd40de99415459ded891331a1a56b

amiclaus, 2018-03-13:
On Scopy build: 2e31a75 (v0.11) the bug is fixed.

closed 2018-03-13 by amiclaus