analogdevicesinc/scopy #361: Labels not uniform when volts/div is set to mV

tagoylo, 2018-02-12:
Scopy build: [9a6660c3](https://github.com/analogdevicesinc/scopy/commit/9a6660c3d64a4ac461944adc750a767a70211d96)
firmware version: v0.18

Steps to reproduce:

check box to enable labels on the plot on general settings
set volts/div: 100mV or 50 mV
position: 0v
the result is shown below
labes are not uniform, others are in mV and others in V

This happens on all channels and math channels.

[image]
[image]

DanielGuramulta, 2018-03-19:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/12e03c54bb74a6d81d56ae2bf649a6c61927443f

amiclaus, 2018-03-22:
On Scopy build: 1beb185 (v0.11) the bug is fixed.

closed 2018-03-22 by amiclaus