analogdevicesinc/scopy #362: Signal Generator: Signal Distorted at low frequency and high voltage

drbautista, 2018-02-12:
OS: Windows 10 64bit
Scopy Version: af505af
Firmware Version: v0.18

Steps to reproduce:
1. Open Signal Generator and set to any of this waveform type (Sine, Triangle, Trapezoidal or sawtooth).
2. Set amplitude to 9V to 10V
3. Set Frequency to 100Hz or below
4. The signal generated is distorted.

Sample:

[image]

Similar distorted signal is generated on Sinewave and Trapezoidal wave

[image]

Sawtooth wave distortion

adisuciu, 2018-02-16:
Cannot reproduce the error. Can you reproduce this on each start ? Does the board calibrate when initializing ? Are you using the latest hardware ?

drbautista, 2018-02-21:
Hi Adrian,

Yes this bug is reproduced each start and the auto calibration is also initiated.
I'm using the latest firmware and scopy version right now but currently using the old board revision. There are new boards coming here and will retest this bug as soon as it gets here.

Thanks a lot.

adisuciu, 2018-03-13:
Retest this with new boards revision.

closed 2018-03-19 by amiclaus