analogdevicesinc/scopy #365: Inaccurate generated Math functions

amiclaus, 2018-02-15:
It seems that for some math functions, the signal is distorted when using decimal numbers inside a trigonometric function.
Here is the result:
[image]

[image]

To be mentioned that for integer numbers, the signal is correct.
[image]
[image]

rgetz, 2018-02-15:
I think this is actually OK - (sort of)

The buffer has n samples - its up to the user to make sure the math in the equation has the same sample at the start of the buffer, and the end of the buffer. It's just a matter of clearly communicating to end users that the n sample in the buffer is (t-n/2) and the end of the buffer is (t+n/2). Or however the mapping is made.

-Robin

amiclaus, 2018-02-16:
In this case, what happens behind the math functions needs to be documented. At a first glance, as an external user, I would expect in both situations to have modifications only on the signal's frequency (and not on the shape).

damercer, 2018-02-19:
The average user is thinking in this order when creating math waveforms, First amplitude, offset, frequency and last ( if at all ) the number of samples needed based on sample rate. The math interface should assume the user wants a continuous waveform and adjust the number of samples ( and / or tweak the frequency a little ) to make that happen.

Putting some examples showing how to do that in the documentation for the signal generator math functions of course will go a long way to help the user get the waveform they want.

Doug

adisuciu, 2018-03-15:
It looks like there is a bug in gnuradio, or am i missing something ?

Inputs:
[image]

Outputs:
[image]

adisuciu, 2020-07-16:
related to https://github.com/analogdevicesinc/scopy/issues/821

adisuciu, 2020-07-29:
This was fixed with #833

closed 2020-07-29 by adisuciu