analogdevicesinc/scopy #367: No indicator for other hidden instruments when menu is collapsed

tagoylo, 2018-02-20:
scopy build: [7f1b98cd](https://github.com/analogdevicesinc/scopy/commit/7f1b98cd84e2cb1d0cb570d214fa946aeb9c73ef)
firmware: v0.19

When menu is collapsed, as shown in the picture below, there is no indicator that there are more instruments hidden. Perhaps an arrow or the scroll bar be present for the user to know that there are still instruments below.

[image]

DanielGuramulta, 2018-03-07:
Fixed with: https://github.com/analogdevicesinc/scopy/commit/3bff072aaa904c89c4db673e8d6105ccb8cb8658

amiclaus, 2018-03-12:
On Scopy build: 6bd7075 (v0.11) the feature is implemented.

closed 2018-03-12 by amiclaus