analogdevicesinc/scopy #368: Upon launch, output of some channel doesn't correspond to what is set by default

tagoylo, 2018-02-20:
scopy build: [7f1b98cd](https://github.com/analogdevicesinc/scopy/commit/7f1b98cd84e2cb1d0cb570d214fa946aeb9c73ef)
firmware: v0.19
 
steps to reproduce

delete first scopy.ini in C:\Users\your_username\AppData\Roaming\ADI 
Connect DIO0 to scope ch1 Go to pattern generator, enable DIO0
Do not change any parameters aside from enabling it, keep it as is
DIO0 is by default in OD output
Then run pattern generator and monitor it in oscilloscope
The result is that DIO0, although output is set to OP, it output as PP. 
The same happens with the following channels set by default as PP but its output is OD: 
DIO2, DIO3, DIO4, DIO5, DIO6, DIO7

It only outputs the correct output once changed to other output option then back to the other. 
OD(default) ->change to PP ->change to OD   or
PP(default) ->change to OD ->change to PP

[image]
[image]

adisuciu, 2018-03-14:
fixed with f92cd91

amiclaus, 2018-03-15:
On Scopy build: 355a04c (v0.11) the bug is fixed.

closed 2018-03-15 by amiclaus