analogdevicesinc/scopy #376: Scopy crashes after clicking connect

tagoylo, 2018-02-22:
Scopy build: [793bd53c](https://github.com/analogdevicesinc/scopy/commit/793bd53c5c71a46a54210bee1176172b398a26d0)
Firmware: v0.19

Click device then click connect
after a few second Sopy crashes.

[image]
[image]

adisuciu, 2018-02-22:
You are testing a pull request. That commit is not merged to master. We are currently investigating the crash.

AlexandraTrifan, 2018-03-01:
The feature implemented in the pull request you were testing is now merged to master and this issue should be fixed after commit 27e453cbeb9606e9bf5884107cff210d3e0fab31.

amiclaus, 2018-03-13:
On Scopy build: 2e31a75 (v0.11) the bug is fixed.

closed 2018-03-13 by amiclaus