analogdevicesinc/scopy #381: Marker jumps out of the current sweep

amiclaus, 2018-03-01:
Scopy version: 23caddb

I.e. Set sweep range from 1MHz to 10MHz. Enable Marker Table. Select and display Marker1 on the plot. Press the "Peak" button to get the marker to the highest detected peak. 

Result: marker is placed in this case (with no external circuity added) to the highest peak detected (the DC component) which is outside the selected sweep range. See picture.

[image]

I think that the markers should work only within the current sweep range.

dNechita, 2018-03-03:
For automatic marker detection (highest peak, next/prev peak and possibly others) it makes sense to apply the marker search on the sweep range. You could even go further and define a "Domain Search" where user can configure the searchable span.
But I wouldn't generalize that markers should only work within the current sweep range as offscreen markers can be useful too.
Note that because of the way the frequency spectrum is currently being computed, the spectrum will always start from 0Hz. Setting the start frequency to higher values will only affect the visible part of the trace (the trace will always contain data starting from 0Hz).

adisuciu, 2018-03-20:
Fixed with  c40394f

amiclaus, 2018-03-22:
On Scopy build: 1beb185 (v0.11) the bug is fixed.

closed 2018-03-22 by amiclaus