analogdevicesinc/scopy #383: Selected marker enhancement

amiclaus, 2018-03-02:
I think that is more intuitive to have the selected marker yellow filled. 
See pic (M2 is selected, M1 is yellow filled).
[image]

adisuciu, 2018-03-06:
Fixed with https://github.com/analogdevicesinc/scopy/commit/54a986c641ff8ed8068452140b92db2a6f9d375b

amiclaus, 2018-03-12:
On Scopy build: 6bd7075 (v0.11) the selected marker is still not yellow filled. (see pic: m2 selected, m1 yellow filled)

[image]

adisuciu, 2018-08-17:
I think the color is a matter of preference. We can review this when we review the UI.

closed 2024-09-24 by adisuciu