analogdevicesinc/scopy #384: Signal Generator: Phase setting in ns, ns, us, ms and s automatically changes.

drbautista, 2018-03-05:
OS: Windows 10 64bit
Scopy Version: 0bc041b
Firmware Version: v0.18

Steps to reproduce:

1. Open Signal Generator
2. Set both channels to Amplitude 5V, Frequency: 1.25MHz, Offset: 0V, Phase: 0 degrees
3. On Channel 1, Set Phase: 400ns [the graphic representation should be similar to 180 degrees phase shift]
4.  Check channel 2's setting and the phase shift will also be set to 400ns.
5. Go back to channel 1's setting and the phase shift will change again.

The attached photo shows the chronological order of the above's step.

[image]

I also tested it on different frequency and different phase setting.

septimiusarca, 2018-03-13:
Fixed with [095ed7b](https://github.com/analogdevicesinc/scopy/pull/392/commits/095ed7b0c046b3ae685a75f9ead6fc543212f516)

amiclaus, 2018-03-15:
Following the steps after a Scopy reset shows that the bug is not fixed.

adisuciu, 2018-03-19:
Should be fixed now : https://github.com/analogdevicesinc/scopy/pull/398

amiclaus, 2018-03-19:
On Scopy build: 1a38185 the bug is fixed.

closed 2018-03-19 by amiclaus