analogdevicesinc/scopy #386: Channels are set as individual but shows as group

tagoylo, 2018-03-05:
Scopy build: [0bc041b0](https://github.com/analogdevicesinc/scopy/commit/0bc041b0c0feae44b79972aaedcbe93480c78678)

Upon launch, open Digital IO instrument. The picture below is what the interface looks like.
It is set as individual channels but the controls are for a group. It will show the individual controls once you click/choose individual again.

[image]

Also, the in/out buttons in group channels do not have the same GUI with the in/out buttons in individual channels.

DanielGuramulta, 2018-03-05:
Bug fixed with commit: https://github.com/analogdevicesinc/scopy/commit/9044ed49d861aebf864a232796c422dcec2eedf0

amiclaus, 2018-03-12:
On Scopy build: 6bd7075 (v0.11) the bug is fixed.

closed 2018-03-12 by amiclaus