analogdevicesinc/scopy #389: When zooming in network analyzer, the labels are messed up

tagoylo, 2018-03-08:
SCopy build: [7ded1074](https://github.com/analogdevicesinc/scopy/commit/7ded1074e114f56244d4137788d6fd54b727cfe8)

As shown in the picture, when zoomed in, the frequency labels are messed up.
The value also shown by the cursor reflects the messed up labels. 

[image]

adisuciu, 2018-03-13:
This is expected behavior, it indicates that the frequency is 260 khz, with a delta of 1.2khz, 3.59khz, etc. However both plots should be zoomed together to maintain cursor consistency.

rgetz, 2018-03-13:
I thought we discussed number of digits? Why is it 260.000000 rather than just 260.00? (Two digits like the other labels).

Plus it seems to read -4kHz? Shouldn't the min be D.C.? (0Hz?) or is that -4kHz from the center? It's difficult to know which is the absolute value vs the relative value on the scale. The font/colours are the same. (The only difference is the +/- on things, which I guess might be enough).

adisuciu, 2018-03-13:
I agree on the number of digits.
It's 4khz from 260 kHz, basically 256 kHz. We could make center label color white in this case, so there's a contrast between the center and the deltas.

This behavior only occurs on excessive levels of zoom, which makes me think whether we really need that much zoom.

claudiupop, 2018-03-16:
With 1f685f983708cd6a0421ed86011b87a88eebf581 the cursor readout widget will display the absolute frequency instead of the relative frequency when zoomed in.

DanielGuramulta, 2018-03-19:
With https://github.com/analogdevicesinc/scopy/commit/4f8c461c6f0fc2d1b9231ca69c8eb833bc4d541d the center label will be white while the delta labels remain the same color as before.

amiclaus, 2018-03-22:
On Scopy build: 1beb185 (v0.11) the bug is fixed.

closed 2018-03-22 by amiclaus