analogdevicesinc/scopy #393: Digital IO window disrupted when detatched

tagoylo, 2018-03-13:
Scopy build: [2e31a757](https://github.com/analogdevicesinc/scopy/commit/2e31a7574540fe2c456235e9800fec482e02ff06)

Open Scopy and stay in minimum window size, detach DIgital IO instrument.
As seen in the picture below, the interface is disrupted.
The digital IO window has no limit in its minimum length thus the disruption. 

[image]

[image]

adisuciu, 2018-03-14:
Fixed with  8f98fbe

amiclaus, 2018-03-15:
On Scopy build: 355a04c (v0.11) the bug is fixed.

closed 2018-03-15 by amiclaus