analogdevicesinc/scopy #394: Inconsistent decrement  in trigger Level and Hysteresis value

tagoylo, 2018-03-13:
Scopy build: [2e31a75](https://github.com/analogdevicesinc/scopy/commit/2e31a7574540fe2c456235e9800fec482e02ff06)

Inconsistent decrement  in trigger Level and Hysteresis value when volts/div is set to 500mV/div and below

steps to reproduce:
set signal generator to the following:
2V 1kHz sine wave
set oscilloscope to the following:
timebase= 200us/div
Volt/div = 500mV/div

Then run both instruments.
Open trigger settings of the oscilloscope.

For the value of level:
Start at 0 mV level, coarse interval
Then click + to increase. It can be observed that it has 500mV interval until 1V. Above 1V, it has a 100mV interval.
Then decrease the value by clicking "-".
You will notice, that as you continue decreasing the value below 1V, it becomes, 900mV, 400mV, -100mV instead of 500mV, 0 mV.

For the value of hysteresis:
increase value until 1V
then decrease, you will notice that instead of decreasing to 950mV, it skips to 900mV.

AlexandraTrifan, 2018-03-14:
Fixed with commit df9d9ee65b053f08b865fcb89085fe6790e23c37.

amiclaus, 2018-03-15:
On Scopy build: 355a04c (v0.11) the bug is fixed.

closed 2018-03-15 by amiclaus