analogdevicesinc/scopy #395: Power supply instrument does not function

tagoylo, 2018-03-13:
Scopy build: [2e31a75](https://github.com/analogdevicesinc/scopy/commit/2e31a7574540fe2c456235e9800fec482e02ff06)

The power supply does not work, both positive and negative

[image]

mirceacaprioru, 2018-03-13:
The power supply now uses calibration attributes. In order to get ensure correct functionality you have to manually copy m2k_calib.ini on the M2K mass storage drive. The file is attached here.

rgetz, 2018-03-13:
Shouldn't either Scopy use some defaults, or the firmware be updated to make sure a file is there? Still smells like an issue to me.

mhennerich, 2018-03-13:
https://github.com/analogdevicesinc/buildroot/commit/01c8805e605a84d072ba2543a8bbf505c21fea88

The firmware was already updated to provide the default if both the user or factory calibration files are missing. I shared a firmware build with Mircea to use for testing until we release an new official build.

rgetz, 2018-03-13:
Ok - sounds great - can you also send that to @tagoylo as well?

mhennerich, 2018-03-13:
done

amiclaus, 2018-03-15:
Since all the [test results](https://pl.web.analog.com/hpt/GlobalAlliancesSharing/_layouts/15/WopiFrame.aspx?sourcedoc=/hpt/GlobalAlliancesSharing/Shared%20Documents/Scopy%20QA/v0.11/power_supply_-_test_case.xlsx&action=default) for the Power Supply (v0.11) passed, I will close the issue.

closed 2018-03-15 by amiclaus