analogdevicesinc/scopy #396: Digital IO: IN and OUT function is reversed

tagoylo, 2018-03-13:
Scopy build: [2e31a75](https://github.com/analogdevicesinc/scopy/commit/2e31a7574540fe2c456235e9800fec482e02ff06)

when the channel is set to IN, what it does is it functions as OUT

when the channel is set as OUT, it functions as IN.

Steps to reproduce:
Connect DIO0 and DI01
Set DIO0 as OUT, and DIO1 as IN
You'll notice that as you run the instrument and change the logic level of DIO0, which is configured as out,  there is no response in DIO1.
But when changing the logic levels of DIO1, configured as IN, there is a response in DIO0.

[image]

adisuciu, 2018-03-13:
Fixed with  5634194

amiclaus, 2018-03-14:
On Scopy build: 1ead1e1 (v0.11) the bug is fixed.

closed 2018-03-14 by amiclaus