analogdevicesinc/scopy #397: Contrast of position and reading when changing min/max freq

tagoylo, 2018-03-14:
Scopy build: [1ead1e18](https://github.com/analogdevicesinc/scopy/commit/1ead1e18eabc2d7cf519fce86775cd3ce9f67413)

Steps to reproduce:
340Hz LPF
run network analyzer with min freq = 10Hz and max freq = 500kHz
Enable cursor and move to -3dB of the trace.
Then change max freq to 50 kHz.

It can be noticed that the position does not correspond to the reading. 
It will only update when the user moves the cursor.
[image]

claudiupop, 2018-03-16:
Fixed with 09b64c88293af101efd530aa1f1ea8555358df69

amiclaus, 2018-03-19:
On Scopy build: 1a38185 the bug is fixed.

closed 2018-03-19 by amiclaus