analogdevicesinc/scopy #401: scopy/M2k hangs randomly when running for a long time

rgetz, 2018-03-23:
When doing various demos at tradeshows, I normally do something lame like connect signal generator to analog input + (- to GND), and show a sine wave, or math function on the screen.

randomly, sometimes an hour, sometimes 4 hours, sometimes 10 minutes, the GUI in the scope stops updating. Clicking on the stop button, and then the run button fixes things, but it happens again.

This has occurred on multiple desktops/laptops, and was occurring on rev A, as well as rev C hardware.

The device itself seems to be fine, as I can still ssh to things, and poke around...

-Robin

mhennerich, 2018-03-23:
I poked around as well. And could trigger it as well. The interesting part was that only the OSC tool stopped updating the enabled FFT plot was still running.

rgetz, 2018-03-23:
Did you determine how to reproduce? or just wait?

adisuciu, 2018-03-27:
It might be something in the gnuradio flow. We will investigate. It sounds like this is an older issue, or .. ?

rgetz, 2018-05-03:
Thanks for tracking this down. I will try things out when things are rebuilt.

rgetz, 2018-05-21:
Closing this since I can't reproduce anymore.

closed 2018-05-21 by rgetz