analogdevicesinc/scopy #402: Scopy crashes when opening preferences

tagoylo, 2018-03-27:
Scopy  build: [1d0a075f](https://github.com/analogdevicesinc/scopy/commit/1d0a075ff196205524604598a4d023249b58adf9)

Steps to reproduce:

Click Connect
Click Disconnect 
Click again Connect
Then open preferences
Scopy crashes few moments after clicking preferences

[image]

adisuciu, 2018-03-27:
Can you provide an .ini file for the build that crashes ? 
On Windows it's in C:\Users\<User>\AppData\Roaming\ADI\Scopy.ini

tagoylo, 2018-03-27:
here it is,

[Scopy.zip](https://github.com/analogdevicesinc/scopy/files/1851259/Scopy.zip)

adisuciu, 2018-03-27:
Also, same path, preferences.ini

tagoylo, 2018-03-27:
[ini_files.zip](https://github.com/analogdevicesinc/scopy/files/1851398/ini_files.zip)

adisuciu, 2018-03-27:
We cannot reproduce. Do you load something before the crash occurs ?

DanielGuramulta, 2018-03-28:
I was able to reproduce the bug by changing the number of periods for the signal generator. Connect -> Disconnect -> Connect and while some tools are calibrating change the number of periods in the preferences menu. Can you try to reproduce the bug using this build: https://ci.appveyor.com/project/analogdevicesinc/scopy/build/1.0.1737/artifacts @tagoylo

tagoylo, 2018-04-02:
I wasn't able to reproduce the bug with the given build.

adisuciu, 2018-04-05:
Should be fixed with e700029

amiclaus, 2018-05-02:
On Scopy build: 87e17a9b the bug is fixed.

closed 2018-05-02 by amiclaus