analogdevicesinc/scopy #403: the interval gets messed up when phase is in seconds

tagoylo, 2018-03-27:
scopy build: [1d0a075f](https://github.com/analogdevicesinc/scopy/commit/1d0a075ff196205524604598a4d023249b58adf9)

Steps to reproduce

Set waveform as following
Sine wave
500Hz frequency
Set phase to 90 degrees
Then change phase unit to seconds
The value should be 500us.
Then click + to increase phase value.
When phase value reaches 1ms, the interval gets messed up.
The value continues to increase but the wheel/value indicator and the display don't correspond to the value.

[image]

amiclaus, 2018-03-27:
That's because when you increase the value, it is always odd (1ms, 11ms, 21ms, etc). This corresponds to N*360deg+180deg. So you will always see the same signal (with 180deg shift).

amiclaus, 2018-03-28:
I will close the issue, since the bug is not valid.

rgetz, 2018-03-28:
If people are getting confused by the UI - it is still an issue, and should be kept open.

amiclaus, 2018-03-29:
A solution would be to limit the interval in the same manner as for degrees/radians. In this particular case, the interval should be between 0ms and 2ms.

adisuciu, 2018-04-05:
Should be fixed with  cce2fe8

amiclaus, 2018-05-02:
On Scopy build: 87e17a9b the bug is fixed.

closed 2018-05-02 by amiclaus