analogdevicesinc/scopy #405: Add Thickness option for other Instruments

amiclaus, 2018-04-02:
A good idea would be to add thickness option for Network Analyzer plots (and maybe also for Spectrum Analyzer).

This will improve the visibility of the plots especially when taking captures that will be used afterwards.

claudiupop, 2018-06-04:
Implemented with https://github.com/analogdevicesinc/scopy/pull/442

amiclaus, 2018-07-31:
Can we do this also for Spectrum Analyzer? It might be helpful especially when the plot scale is in Volts (Vpeak, Vrms)

claudiupop, 2018-08-06:
On spectrum analyzer, if we set a higher thickness for the curve, the application will work very slowly because there are too many points to be rendered. We will think for a solution to avoid this.

HRosete, 2020-10-21:
This is implemented in Scopy v1.2 (verified on 33ca72f9)on Network Analyzer. Closing this item.

closed 2020-10-21 by HRosete